Xilinx/dma_ip_drivers #265: And here is complete guide for repaire the issue:

mayureshw, 2024-03-20:
And here is complete guide for repaire the issue:
1. File cdev_sgdma.c, function bodies for `cdev_write_iter` and `cdev_read_iter`, need to change `io->iov` to `io->__iov`;
2. File xdma_cdev.c, line 606, change call of `class_create(THIS_MODULE, XDMA_NODE_NAME)` to `class_create(XDMA_NODE_NAME)`;
3. File cdev_ctrl.c, line 236. Need to add conversion of `vma->vm_flags` to `vm_area_struct *`. For example, you could write a function to write a flag into `vm_flags` field of vma (now is const):

```C
// Source cpied from another devforum 
void vm_flags_set(struct vm_area_struct *vma, vm_flags_t flags) {
    vma->vm_flags |= flags;
}
```

Usage:
```C
// ... bridge mmap function body
vm_flags_set(vma, VMEM_FLAGS)
// ... 
```

With this changes, driver completly builds, again :)

_Originally posted by @MADMechaniculus in https://github.com/Xilinx/dma_ip_drivers/issues/253#issuecomment-1897509289_

j-bo, 2024-03-21:
Seems that vm_flags_set has been included elsewhere thus causing your error. 

Remove the vm_flags_set() definition you added and simply call existing ` vm_flags_set(vma, VMEM_FLAGS);` (do not forget the ';')

Works on my setup (Ubuntu22.04 kernel 6.5.0-1018-oem)

mayureshw, 2024-03-27:
Right. Thanks.

closed 2024-03-27 by mayureshw