Xilinx/dma_ip_drivers #267: assignment of read-only member ‘vm_flags’

weidingxin3, 2024-03-22:
My system:kubuntu 23.10
linux kernel:6.5.0-26-generic
when I cd to dma_ip_drivers/XDMA/linux-kernel/xdma
and make.It show dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_ctrl.c:236:23: error: assignment of read-only member ‘vm_flags’
  236 |         vma->vm_flags |= VMEM_FLAGS;

How can i do?
Thanke you for any help.

gonsolo, 2024-04-25:
Update drivers: https://github.com/Xilinx/dma_ip_drivers/pull/275

Unlicensed-driver-ljx, 2024-09-14:
如何更新这个程序呢

Unlicensed-driver-ljx, 2024-09-14:
Makefile:17: XVC_FLAGS: .
make -C /lib/modules/6.1.57/build M=/home/pi/dma_ip_drivers/XDMA/linux-kernel/xdma modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.57'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: aarch64-linux-gcc (ctng-1.25.0-119g-FA) 11.3.0
  You are using:           gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
/home/pi/dma_ip_drivers/XDMA/linux-kernel/xdma/Makefile:17: XVC_FLAGS: .
  CC [M]  /home/pi/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.o
  CC [M]  /home/pi/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.o
  CC [M]  /home/pi/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_ctrl.o
/home/pi/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_ctrl.c: In function 'bridge_mmap':
/home/pi/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_ctrl.c:236:23: error: assignment of read-only member 'vm_flags'
  236 |         vma->vm_flags |= VMEM_FLAGS;
      |                       ^~
make[2]: *** [scripts/Makefile.build:250: /home/pi/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_ctrl.o] Error 1
make[1]: *** [Makefile:2057: /home/pi/dma_ip_drivers/XDMA/linux-kernel/xdma] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.57'
make: *** [Makefile:39: all] Error 2

hinxx, 2024-09-14:
See PR #292. 

Sadly Xilinx/AMD is not merging any of code from PRs hence the issues with new kernels.