Xilinx/dma_ip_drivers #269: The xdma linux kernel driver does not build against Ubuntu kernel 6.5.x (Ubuntu 22.04.1_LTS). Any idea?

tw00000, 2024-04-02:
Makefile:17: XVC_FLAGS: .

make -C /lib/modules/6.5.0-26-generic/build M=/home/truss/xilinx/dma_ip_drivers/XDMA/linux-kernel/xdma modules

make[1]: Entering directory '/usr/src/linux-headers-6.5.0-26-generic'

warning: the compiler differs from the one used to build the kernel

  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0

  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0

/home/truss/xilinx/dma_ip_drivers/XDMA/linux-kernel/xdma/Makefile:17: XVC_FLAGS: .

  CC [M]  /home/truss/xilinx/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.o

  CC [M]  /home/truss/xilinx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.o

In file included from ./include/linux/linkage.h:7,

                 from ./include/linux/kernel.h:17,

                 from /home/truss/xilinx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.h:23,

                 from /home/truss/xilinx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:22:

/home/truss/xilinx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c: In function ‘xdma_cdev_init’:

./include/linux/export.h:29:22: error: passing argument 1 of ‘class_create’ from incompatible pointer type [-Werror=incompatible-pointer-types]

   29 | #define THIS_MODULE (&__this_module)

      |                     ~^~~~~~~~~~~~~~~

      |                      |

      |                      struct module *

/home/truss/xilinx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:606:37: note: in expansion of macro ‘THIS_MODULE’

  606 |         g_xdma_class = class_create(THIS_MODULE, XDMA_NODE_NAME);

      |                                     ^~~~~~~~~~~

In file included from ./include/linux/device.h:31,

                 from ./include/linux/cdev.h:8,

                 from /home/truss/xilinx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_mod.h:25,

                 from /home/truss/xilinx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.h:27:

./include/linux/device/class.h:230:54: note: expected ‘const char *’ but argument is of type ‘struct module *’

  230 | struct class * __must_check class_create(const char *name);

      |                                          ~~~~~~~~~~~~^~~~

/home/truss/xilinx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:606:24: error: too many arguments to function ‘class_create’

  606 |         g_xdma_class = class_create(THIS_MODULE, XDMA_NODE_NAME);

      |                        ^~~~~~~~~~~~

./include/linux/device/class.h:230:29: note: declared here

  230 | struct class * __must_check class_create(const char *name);

      |                             ^~~~~~~~~~~~

cc1: some warnings being treated as errors

make[3]: *** [scripts/Makefile.build:251: /home/truss/xilinx/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.o] Error 1

make[2]: *** [/usr/src/linux-headers-6.5.0-26-generic/Makefile:2039: /home/truss/xilinx/dma_ip_drivers/XDMA/linux-kernel/xdma] Error 2

make[1]: *** [Makefile:234: __sub-make] Error 2

make[1]: Leaving directory '/usr/src/linux-headers-6.5.0-26-generic'

make: *** [Makefile:39: all] Error 2

MischaBaars, 2024-04-02:
Try this one: https://github.com/MischaBaars/dma_ip_drivers. The XDMA and QDMA parts will compile for the newer kernels. The XVSEC part won't compile for the newer kernels yet. That part makes use of structures (mm_segment_t) for which kernel support has been removed as of kernel 5.17.16. It probably does compile for the 5.15.x kernels you're using.

Then again, I'm just getting started as you are and I'm still looking into the details of the XDMA part. I have no idea what the XVSEC part does.

gonsolo, 2024-04-25:
I got xdma and xvsec working for 6.7.7: https://github.com/Xilinx/dma_ip_drivers/pull/275