Xilinx/dma_ip_drivers #287: XDMA driver for ubuntu24.04

ate-wang, 2024-07-08:
Can you add the xdma driver for ubuntu24.04? Can you update the version of ubuntu?

ate-wang, 2024-08-02:
I see your recovery is very good, can I ask you a question? I need to compile the xdma driver in the Kylin V10 system. Has anyone adapted the xdma driver in Kylin? I have a problem now. When the host computer operates pcie xdma, the system will freeze. After checking the log, I found that pcie is stuck. What is the situation when pcie (pcie ff) is stuck? Have you encountered this before?

hinxx, 2024-08-19:
> Can you add the xdma driver for ubuntu24.04? Can you update the version of ubuntu?

The driver code can by used with any flavor/distribution of the Linux that has kernel version at least >= 2.6.

> I need to compile the xdma driver in the Kylin V10 system. Has anyone adapted the xdma driver in Kylin?

If this is Linux kernel based OS you should expect it to work, otherwise someone needs to do a proper port.