Xilinx/dma_ip_drivers #288: Building failed

notooth1, 2024-07-11:
I got some errors when building this package on AlmaLinux 9.4. Please help.
```
$ git clone https://github.com/Xilinx/dma_ip_drivers

$ cd dma_ip_drivers/XDMA/linux-kernel/xdma/

$ make
bash: cd: ../../XDMA/linux-kernel/xdma/: No such file or directory
notooth@localhost:xdma$ make
Makefile:17: XVC_FLAGS: .
make -C /lib/modules/5.14.0-427.22.1.el9_4.x86_64/build M=/mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma modules
make[1]: Entering directory '/usr/src/kernels/5.14.0-427.22.1.el9_4.x86_64'
/mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/Makefile:17: XVC_FLAGS: .
  CC [M]  /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/libxdma.o
  CC [M]  /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.o
In file included from ./include/linux/linkage.h:7,
                 from ./include/linux/kernel.h:8,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.h:23,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:22:
/mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c: In function ‘xdma_cdev_init’:
./include/linux/export.h:17:22: error: passing argument 1 of ‘class_create’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   17 | #define THIS_MODULE (&__this_module)
      |                     ~^~~~~~~~~~~~~~~
      |                      |
      |                      struct module *
/mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:606:37: note: in expansion of macro ‘THIS_MODULE’
  606 |         g_xdma_class = class_create(THIS_MODULE, XDMA_NODE_NAME);
      |                                     ^~~~~~~~~~~
In file included from ./include/linux/device.h:31,
                 from ./include/linux/cdev.h:8,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_mod.h:25,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.h:27,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:22:
./include/linux/device/class.h:230:54: note: expected ‘const char *’ but argument is of type ‘struct module *’
  230 | struct class * __must_check class_create(const char *name);
      |                                          ~~~~~~~~~~~~^~~~
/mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:606:24: error: too many arguments to function ‘class_create’
  606 |         g_xdma_class = class_create(THIS_MODULE, XDMA_NODE_NAME);
      |                        ^~~~~~~~~~~~
In file included from ./include/linux/device.h:31,
                 from ./include/linux/cdev.h:8,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_mod.h:25,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.h:27,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:22:
./include/linux/device/class.h:230:29: note: declared here
  230 | struct class * __must_check class_create(const char *name);
      |                             ^~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:299: /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.o] Error 1
make[1]: *** [Makefile:1934: /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.0-427.22.1.el9_4.x86_64'
make: *** [Makefile:39: all] Error 2
```

hinxx, 2024-07-11:
Take a look at https://github.com/Xilinx/dma_ip_drivers/issues/262. You will need to change the code after clone yourself..

notooth1, 2024-07-12:
After following this [commit](https://github.com/Xilinx/dma_ip_drivers/pull/240/commits/594e39045fb032de90150e7fae0ec969eba9d8d5) to change the code, I still got some errors:
```
$ make
Makefile:17: XVC_FLAGS: .
make -C /lib/modules/5.14.0-427.22.1.el9_4.x86_64/build M=/mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma modules
make[1]: Entering directory '/usr/src/kernels/5.14.0-427.22.1.el9_4.x86_64'
/mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/Makefile:17: XVC_FLAGS: .
  CC [M]  /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.o
In file included from ./include/linux/linkage.h:7,
                 from ./include/linux/kernel.h:8,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.h:23,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:22:
/mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c: In function ‘xdma_cdev_init’:
./include/linux/export.h:17:22: error: passing argument 1 of ‘class_create’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   17 | #define THIS_MODULE (&__this_module)
      |                     ~^~~~~~~~~~~~~~~
      |                      |
      |                      struct module *
/mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:607:37: note: in expansion of macro ‘THIS_MODULE’
  607 |         g_xdma_class = class_create(THIS_MODULE, XDMA_NODE_NAME);
      |                                     ^~~~~~~~~~~
In file included from ./include/linux/device.h:31,
                 from ./include/linux/cdev.h:8,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_mod.h:25,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.h:27,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:22:
./include/linux/device/class.h:230:54: note: expected ‘const char *’ but argument is of type ‘struct module *’
  230 | struct class * __must_check class_create(const char *name);
      |                                          ~~~~~~~~~~~~^~~~
/mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:607:24: error: too many arguments to function ‘class_create’
  607 |         g_xdma_class = class_create(THIS_MODULE, XDMA_NODE_NAME);
      |                        ^~~~~~~~~~~~
In file included from ./include/linux/device.h:31,
                 from ./include/linux/cdev.h:8,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_mod.h:25,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.h:27,
                 from /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.c:22:
./include/linux/device/class.h:230:29: note: declared here
  230 | struct class * __must_check class_create(const char *name);
      |                             ^~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:299: /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.o] Error 1
make[1]: *** [Makefile:1934: /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.0-427.22.1.el9_4.x86_64'
make: *** [Makefile:39: all] Error 2
```

hinxx, 2024-07-12:
If you used that commit as-is its code changes are likely not kicking in because they pertain kernel versions 6.x and are guarded by the #defines. Looking at your compile output you are running 5.14 (with backported code from 6.x it seems).

Try removing the #defines from the commit such that the code for 6.x remains.

notooth1, 2024-07-12:
I do not see the #defines in the commit.

hinxx, 2024-07-13:
Sorry, I meant #if/#ifdef blocks.

notooth1, 2024-07-13:
I manually added the code changes and got some errors:
```
$ make
Makefile:17: XVC_FLAGS: .
make -C /lib/modules/5.14.0-427.22.1.el9_4.x86_64/build M=/mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma modules
make[1]: Entering directory '/usr/src/kernels/5.14.0-427.22.1.el9_4.x86_64'
/mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/Makefile:17: XVC_FLAGS: .
  CC [M]  /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.o
  CC [M]  /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_ctrl.o
/mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_ctrl.c: In function ‘bridge_mmap’:
/mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_ctrl.c:239:9: error: implicit declaration of function ‘vm_flags_set’ [-Werror=implicit-function-declaration]
  239 |         vm_flags_set(vma, VMEM_FLAGS);
      |         ^~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:299: /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_ctrl.o] Error 1
make[1]: *** [Makefile:1934: /mnt/Archive/Downloads/1/dma_ip_drivers/XDMA/linux-kernel/xdma] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.0-427.22.1.el9_4.x86_64'
make: *** [Makefile:39: all] Error 2
```

hinxx, 2024-07-13:
Progress. 

If you take a look at the same commit, where you took the code from, there are couple of more files you might need to change.

notooth1, 2024-07-14:
Is there a patch file to apply all the commits? I am unable to change all the commit code manually.

hinxx, 2024-07-14:
There are 4-5 lines of code you need to change across three files. I think that is a really small patch to do manually. 

Nevertheless, this post https://stackoverflow.com/questions/6188591/download-github-pull-request-as-unified-diff#6188624 explains how to get a diff/patch file from a PR or commit on github. Beware that a patch against that 6.5 version might not apply to yours and then you're back to manual work.

notooth1, 2024-07-15:
I followed the post, but it did not work:
```
$ git checkout -b master
fatal: a branch named 'master' already exists

$ git pull origin pull/240/head
From https://github.com/Xilinx/dma_ip_drivers
 * branch            refs/pull/240/head -> FETCH_HEAD
hint: You have divergent branches and need to specify how to reconcile them.
hint: You can do so by running one of the following commands sometime before
hint: your next pull:
hint: 
hint:   git config pull.rebase false  # merge
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
fatal: Need to specify how to reconcile divergent branches.I downloaded the master branch of https://github.com/Xilinx/dma_ip_drivers. Now I want to apply this commit https://github.com/Xilinx/dma_ip_drivers/pull/240
```

hinxx, 2024-07-15:
I thougt you were after the patch out of commit. That is what the link I provided shows
[image]

notooth1, 2024-07-15:
I still got this error:
```
$ git apply --stat 240.patch
 XDMA/linux-kernel/xdma/cdev_ctrl.c  |    5 ++++
 XDMA/linux-kernel/xdma/cdev_sgdma.c |   14 ++++++++++++
 XDMA/linux-kernel/xdma/xdma_cdev.c  |    4 ++++
 XDMA/linux-kernel/xdma/xdma_mod.c   |    4 ++--
 XDMA/linux-kernel/xdma/cdev_ctrl.c  |    4 ++--
 XDMA/linux-kernel/xdma/libxdma.c    |   34 ++++++++++++++++--------------
 XDMA/linux-kernel/xdma/cdev_sgdma.c |   25 ++++++++++------------
 XDMA/linux-kernel/xdma/xdma_mod.c   |    4 ++--
 XDMA/linux-kernel/xdma/xdma_mod.h   |    4 ++--
 XDMA/linux-kernel/xdma/libxdma.c    |    7 +-----
 XDMA/linux-kernel/xdma/cdev_ctrl.c  |    4 ++--
 XDMA/linux-kernel/xdma/libxdma.c    |    6 +++--
 XDMA/linux-kernel/xdma/libxdma.c    |    1 +
 XDMA/linux-kernel/tools/Makefile    |   40 ++++++++++++++++++++++++-----------
 XDMA/linux-kernel/xdma/Makefile     |   22 +++++++++++--------
 XDMA/linux-kernel/xdma/Makefile     |    2 +-
 XDMA/linux-kernel/xdma/xdma_mod.c   |    2 +-
 17 files changed, 109 insertions(+), 73 deletions(-)

$ git apply --check 240.patch
 
$ git apply 240.patch

$ cd XDMA/linux-kernel/xdma/

$ make
Makefile:17: XVC_FLAGS: .
make -C /lib/modules/5.14.0-427.24.1.el9_4.x86_64/build M=/mnt/Archive/Downloads/4/dma_ip_drivers/XDMA/linux-kernel/xdma modules
make[1]: Entering directory '/usr/src/kernels/5.14.0-427.24.1.el9_4.x86_64'
/mnt/Archive/Downloads/4/dma_ip_drivers/XDMA/linux-kernel/xdma/Makefile:17: XVC_FLAGS: .
  CC [M]  /mnt/Archive/Downloads/4/dma_ip_drivers/XDMA/linux-kernel/xdma/xdma_cdev.o
  CC [M]  /mnt/Archive/Downloads/4/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_ctrl.o
/mnt/Archive/Downloads/4/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_ctrl.c: In function ‘bridge_mmap’:
/mnt/Archive/Downloads/4/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_ctrl.c:239:9: error: implicit declaration of function ‘vm_flags_set’ [-Werror=implicit-function-declaration]
  239 |         vm_flags_set(vma, VMEM_FLAGS);
      |         ^~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:299: /mnt/Archive/Downloads/4/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_ctrl.o] Error 1
make[1]: *** [Makefile:1934: /mnt/Archive/Downloads/4/dma_ip_drivers/XDMA/linux-kernel/xdma] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.0-427.24.1.el9_4.x86_64'
make: *** [Makefile:41: all] Error 2
```

hinxx, 2024-07-16:
Failing at this line:

>   239 |         vm_flags_set(vma, VMEM_FLAGS);

makes me think that this part of the 6.x kernel code was not not backported to your kernel version. What happens if you do not include this change from the commit?

In other words, keep the original line:

```
vma->vm_flags |= VMEM_FLAGS;
```

notooth1, 2024-07-16:
This fixed the issue: `vma->vm_flags |= VMEM_FLAGS;`
and I faced other errors:
```
/mnt/Archive/Downloads/5/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_sgdma.c: In function ‘async_io_handler’:
/mnt/Archive/Downloads/5/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_sgdma.c:109:17: error: too many arguments to function ‘caio->iocb->ki_complete’
  109 |                 caio->iocb->ki_complete(caio->iocb, res, res2);
      |                 ^~~~
/mnt/Archive/Downloads/5/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_sgdma.c:126:9: error: too many arguments to function ‘caio->iocb->ki_complete’
  126 |         caio->iocb->ki_complete(caio->iocb, numbytes, -EBUSY);
      |         ^~~~
/mnt/Archive/Downloads/5/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_sgdma.c: In function ‘cdev_write_iter’:
/mnt/Archive/Downloads/5/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_sgdma.c:569:41: error: ‘struct iov_iter’ has no member named ‘iov’; did you mean ‘__iov’?
  569 |         return cdev_aio_write(iocb, io->iov, io->nr_segs, io->iov_offset);
      |                                         ^~~
      |                                         __iov
/mnt/Archive/Downloads/5/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_sgdma.c: In function ‘cdev_read_iter’:
/mnt/Archive/Downloads/5/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_sgdma.c:574:40: error: ‘struct iov_iter’ has no member named ‘iov’; did you mean ‘__iov’?
  574 |         return cdev_aio_read(iocb, io->iov, io->nr_segs, io->iov_offset);
      |                                        ^~~
      |                                        __iov
/mnt/Archive/Downloads/5/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_sgdma.c: In function ‘cdev_write_iter’:
/mnt/Archive/Downloads/5/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_sgdma.c:570:1: error: control reaches end of non-void function [-Werror=return-type]
  570 | }
      | ^
/mnt/Archive/Downloads/5/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_sgdma.c: In function ‘cdev_read_iter’:
/mnt/Archive/Downloads/5/dma_ip_drivers/XDMA/linux-kernel/xdma/cdev_sgdma.c:575:1: error: control reaches end of non-void function [-Werror=return-type]
  575 | }
      | ^
```

hinxx, 2024-07-16:
I guess what you're seeing is similar to issues observed as so far:
a) the kernel 6.x code was ported to your kernel (you need to take the code changes in the PR/commit), or 
b) the kernel 6.x code was NOT ported to your kernel (you need to keep the original code)

Hard to say which is it going to be, case by case, but trial and error approach, as performed so far, should work for the outstanding errors. Good luck!

ate-wang, 2024-08-02:
I see your recovery is very good, can I ask you a question? I need to compile the xdma driver in the Kylin V10 system. Has anyone adapted the xdma driver in Kylin? I have a problem now. When the host computer operates pcie xdma, the system will freeze. After checking the log, I found that pcie is stuck. What is the situation when pcie (pcie ff) is stuck? Have you encountered this before?

akgrummer, 2025-02-13:
Using AlmaLInux9.5, 
and `/usr/src/kernels/5.14.0-503.22.1.el9_5.x86_64`
`make` failed at this [line](https://github.com/Xilinx/dma_ip_drivers/blob/31a4a12ad30fd1aaf1567ed1396f08c24b9ed7b2/XDMA/linux-kernel/xdma/cdev_ctrl.c#L239) now: 
```
239. vma->vm_flags |= VMEM_FLAGS;
```
using [this](https://github.com/Xilinx/dma_ip_drivers/blob/31a4a12ad30fd1aaf1567ed1396f08c24b9ed7b2/XDMA/linux-kernel/xdma/cdev_ctrl.c#L237) instead: `vm_flags_set(vma, VMEM_FLAGS);` 
seems to have worked.