Xilinx/dma_ip_drivers #298: The XDMA did not work when I send big buffer

GitSoftwareNow, 2024-11-07:
[ 1862.197073] libxdma:xdma_xfer_submit: xfer 0x000000009e11f7b1,2048, s 0x1 timed out, ep 0x800.
[ 1862.197096] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0x1fc18006 (id).
[ 1862.197110] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x000000007e34bd2e) = 0x00000001 (status).
[ 1862.197122] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000971a729a) = 0x00f83e1f (control)
[ 1862.197135] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x000000002eee633a) = 0xef0701c0 (first_desc_lo)
[ 1862.197147] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000701d88a0) = 0x00000000 (first_desc_hi)
[ 1862.197159] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x000000004e7cddaf) = 0x00000001 (first_desc_adjacent).
[ 1862.197172] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000d5b48b63) = 0x00000000 (completed_desc_count).
[ 1862.197184] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x000000004c22db83) = 0x00f83e1e (interrupt_enable_mask)
[ 1862.197200] libxdma:engine_status_dump: SG engine 0-C2H0-ST status: 0x00000001: BUSY
[ 1862.197213] libxdma:transfer_abort: abort transfer 0x000000009e11f7b1, desc 2, engine desc queued 0.
[ 1871.183753] libxdma:xdma_xfer_submit: xfer 0x000000009e11f7b1,2048, s 0x1 timed out, ep 0x800.
[ 1871.183777] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0x1fc18006 (id).
[ 1871.183791] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x000000007e34bd2e) = 0x00000001 (status).
[ 1871.183803] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000971a729a) = 0x00f83e1f (control)
[ 1871.183816] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x000000002eee633a) = 0xef070200 (first_desc_lo)
[ 1871.183828] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000701d88a0) = 0x00000000 (first_desc_hi)
[ 1871.183840] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x000000004e7cddaf) = 0x00000001 (first_desc_adjacent).
[ 1871.183853] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000d5b48b63) = 0x00000000 (completed_desc_count).
[ 1871.183865] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x000000004c22db83) = 0x00f83e1e (interrupt_enable_mask)
[ 1871.183882] libxdma:engine_status_dump: SG engine 0-C2H0-ST status: 0x00000001: BUSY
[ 1871.183895] libxdma:transfer_abort: abort transfer 0x000000009e11f7b1, desc 2, engine desc queued 0.
[ 1881.600961] libxdma:xdma_xfer_submit: xfer 0x000000009e11f7b1,2048, s 0x1 timed out, ep 0x800.
[ 1881.600986] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0x1fc18006 (id).
[ 1881.600999] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x000000007e34bd2e) = 0x00000001 (status).
[ 1881.601012] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000971a729a) = 0x00f83e1f (control)
[ 1881.601025] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x000000002eee633a) = 0xef070240 (first_desc_lo)
[ 1881.601037] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000701d88a0) = 0x00000000 (first_desc_hi)
[ 1881.601049] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x000000004e7cddaf) = 0x00000000 (first_desc_adjacent).
[ 1881.601061] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000d5b48b63) = 0x00000000 (completed_desc_count).
[ 1881.601073] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x000000004c22db83) = 0x00f83e1e (interrupt_enable_mask)
[ 1881.601089] libxdma:engine_status_dump: SG engine 0-C2H0-ST status: 0x00000001: BUSY
[ 1881.601103] libxdma:transfer_abort: abort transfer 0x000000009e11f7b1, desc 1, engine desc queued 0.
[ 3284.527090] libxdma:xdma_xfer_submit: xfer 0x0000000001a74a2f,2048, s 0x1 timed out, ep 0x800.
[ 3284.539692] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0xffffffff (id).
[ 3284.539696] libxdma:engine_reg_dump: 0-C2H0-ST: engine id missing, 0xfff00000 exp. & 0xfff00000 = 0x1fc00000
[ 3284.539720] libxdma:engine_status_read: Failed to dump register
[ 3284.539727] libxdma:xdma_xfer_submit: Failed to read engine status
[ 3294.553913] libxdma:xdma_xfer_submit: xfer 0x0000000001a74a2f,2048, s 0x1 timed out, ep 0x800.
[ 3294.566516] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0xffffffff (id).
[ 3294.566519] libxdma:engine_reg_dump: 0-C2H0-ST: engine id missing, 0xfff00000 exp. & 0xfff00000 = 0x1fc00000
[ 3294.566544] libxdma:engine_status_read: Failed to dump register
[ 3294.566552] libxdma:xdma_xfer_submit: Failed to read engine status
[ 3304.580948] libxdma:xdma_xfer_submit: xfer 0x0000000001a74a2f,2048, s 0x1 timed out, ep 0x800.
[ 3304.593552] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0xffffffff (id).
[ 3304.593557] libxdma:engine_reg_dump: 0-C2H0-ST: engine id missing, 0xfff00000 exp. & 0xfff00000 = 0x1fc00000
[ 3304.593586] libxdma:engine_status_read: Failed to dump register
[ 3304.593595] libxdma:xdma_xfer_submit: Failed to read engine status
[ 3314.607738] libxdma:xdma_xfer_submit: xfer 0x0000000001a74a2f,2048, s 0x1 timed out, ep 0x800.
[ 3314.620344] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0xffffffff (id).
[ 3314.620351] libxdma:engine_reg_dump: 0-C2H0-ST: engine id missing, 0xfff00000 exp. & 0xfff00000 = 0x1fc00000
[ 3314.620387] libxdma:engine_status_read: Failed to dump register
[ 3314.620399] libxdma:xdma_xfer_submit: Failed to read engine status
[ 3324.634758] libxdma:xdma_xfer_submit: xfer 0x0000000001a74a2f,2048, s 0x1 timed out, ep 0x800.
[ 3324.647362] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0xffffffff (id).
[ 3324.647369] libxdma:engine_reg_dump: 0-C2H0-ST: engine id missing, 0xfff00000 exp. & 0xfff00000 = 0x1fc00000
[ 3324.647406] libxdma:engine_status_read: Failed to dump register
[ 3324.647418] libxdma:xdma_xfer_submit: Failed to read engine status
[ 3334.661953] libxdma:xdma_xfer_submit: xfer 0x0000000001a74a2f,2048, s 0x1 timed out, ep 0x800.
[ 3334.674555] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0xffffffff (id).
[ 3334.674565] libxdma:engine_reg_dump: 0-C2H0-ST: engine id missing, 0xfff00000 exp. & 0xfff00000 = 0x1fc00000
[ 3334.674613] libxdma:engine_status_read: Failed to dump register
[ 3334.674628] libxdma:xdma_xfer_submit: Failed to read engine status
[ 3344.688782] libxdma:xdma_xfer_submit: xfer 0x0000000000d859e5,2048, s 0x1 timed out, ep 0x800.
[ 3344.701384] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0xffffffff (id).
[ 3344.701393] libxdma:engine_reg_dump: 0-C2H0-ST: engine id missing, 0xfff00000 exp. & 0xfff00000 = 0x1fc00000
[ 3344.701442] libxdma:engine_status_read: Failed to dump register
[ 3344.701460] libxdma:xdma_xfer_submit: Failed to read engine status
[ 3354.715445] libxdma:xdma_xfer_submit: xfer 0x0000000000d859e5,2048, s 0x1 timed out, ep 0x800.
[ 3354.728052] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0xffffffff (id).
[... 78 lines cut ...]
[ 3515.145955] libxdma:xdma_xfer_submit: xfer 0x00000000981075c0,2048, s 0x1 timed out, ep 0x800.
[ 3515.158564] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0xffffffff (id).
[ 3515.158578] libxdma:engine_reg_dump: 0-C2H0-ST: engine id missing, 0xfff00000 exp. & 0xfff00000 = 0x1fc00000
[ 3515.158642] libxdma:engine_status_read: Failed to dump register
[ 3515.158666] libxdma:xdma_xfer_submit: Failed to read engine status
[ 3525.172873] libxdma:xdma_xfer_submit: xfer 0x00000000981075c0,2048, s 0x1 timed out, ep 0x800.
[ 3525.185481] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0xffffffff (id).
[ 3525.185494] libxdma:engine_reg_dump: 0-C2H0-ST: engine id missing, 0xfff00000 exp. & 0xfff00000 = 0x1fc00000
[ 3525.185559] libxdma:engine_status_read: Failed to dump register
[ 3525.185582] libxdma:xdma_xfer_submit: Failed to read engine status
[ 3535.199790] libxdma:xdma_xfer_submit: xfer 0x00000000981075c0,2048, s 0x1 timed out, ep 0x800.
[ 3535.212397] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0xffffffff (id).
[ 3535.212410] libxdma:engine_reg_dump: 0-C2H0-ST: engine id missing, 0xfff00000 exp. & 0xfff00000 = 0x1fc00000
[ 3535.212471] libxdma:engine_status_read: Failed to dump register
[ 3535.212497] libxdma:xdma_xfer_submit: Failed to read engine status
[ 3545.226786] libxdma:xdma_xfer_submit: xfer 0x00000000981075c0,2048, s 0x1 timed out, ep 0x800.
[ 3545.239393] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0xffffffff (id).
[ 3545.239407] libxdma:engine_reg_dump: 0-C2H0-ST: engine id missing, 0xfff00000 exp. & 0xfff00000 = 0x1fc00000
[... cut ...]
[ 3545.239498] libxdma:xdma_xfer_submit: Failed to read engine status
[ 3555.253599] libxdma:xdma_xfer_submit: xfer 0x00000000981075c0,2048, s 0x1 timed out, ep 0x800.
[ 3555.266206] libxdma:engine_reg_dump: 0-C2H0-ST: ioread32(0x00000000866b043a) = 0xffffffff (id).
[ 3555.266221] libxdma:engine_reg_dump: 0-C2H0-ST: engine id missing, 0xfff00000 exp. & 0xfff00000 = 0x1fc00000
[ 3555.266286] libxdma:engine_status_read: Failed to dump register
[ 3555.266309] libxdma:xdma_xfer_submit: Failed to read engine status

mabl, 2024-11-12:
Are you using streams and `TLAST`? There are some bugs with it, that generate unexpected TLAST during transmission. You may try https://github.com/Xilinx/dma_ip_drivers/issues/91#issuecomment-2316828753.

dmitrym1, 2025-01-17:
First three xdma_xfer_submit timeouts you see are probably because of too small timeout value (compared to transaction size, which is big as you mentioned) or because of #311. It's impossible to say from logs which one is your case. Increase the timeout value and/or pull the attached to #311 fix to solve this part of issue.
Next timeouts followed by 0xffffffff are because of #317. Update your Vivado to 2019.1 or newer and upgrade XDMA IP to solve this part of issue.

I also should mention that I'm successfully using XDMA from 2021.1 with @alonbl's patch set in AXI Stream mode and have no problems for more than a year now. So with all available patches it is rock solid. Good luck with your project!