Xilinx/dma_ip_drivers #323: Difference between this driver and the driver in the linux kernel

free43, 2025-02-20:
Hello,

I came across the driver in the Linux kernel (https://github.com/torvalds/linux/blob/master/drivers/dma/xilinx/xdma.c) and Im wondering whats the difference between these two. The one in the kernel is a platform driver, which uses a different interface to load the driver (matching over device trees). 
But what about the implementation differences? Do they work the same way or are they for different  usecases? Are there related reference, which maybe answer my questions (I dont came across something in my research)?

Thanks in advance!

dmitrym1, 2025-02-21:
Hi @free43 , that's an interesting question. In my understanding, the driver in /drivers/dma/xilinx is a _driver for a SoC that has built-in hard IP from Xilinx_, that's why it is instantiated in device tree, just like any other driver for SoC. There is a file that mentions ZynqMP, I guess that's what the driver for. As an opposite, XDMA driver in dma_ip_drivers repo (this one) is _for PCIe device (FPGA with XDMA IP)_ that could be connected to your SoC's PCIe bus.

m2es3h, 2025-12-15:
The driver in the linux kernel adds support for using the Xilinx XDMA IP block with the kernel's DMA Engine API: https://docs.kernel.org/driver-api/dmaengine/client.html

To use it, you would write your own driver to configure this "platform device", request DMA channels by name, then use the kernel's DMA engine API to do the transfers. See this driver (also in the official linux source) for an example of how to do that: https://docs.kernel.org/admin-guide/media/mgb4.html