Xilinx/dma_ip_drivers #330: mmap of a bar fails with EINVAL on kernel 6.6

michkrom, 2025-03-25:
Hi,

I am porting a yokto system from 5.5 kernel to 6.6 so I had to upgrade the XDMA driver from 2019.2.51 to latest. Now the mmap() to resource0 (bar0 but really any bar) fails with EINVAL.

The `mmap()` to bar0 is using /sys/devices/.../resource0. It works normally on old os/xdma bot not the new giving EINVAL. This is only after loading the xdma module - it is fine if the module is not loaded. There is no log output from the driver for mmap() call (added kprintf into bridge_mmap and build it with DEBUG). mmap() via /dev/xdma_.... works and produces log output. This is also reproducible with a standard desktop linux OS (used Manjaro). For testing I used the tools/reg_rw .

I think it is some change in kernel or maybe driver that prevents normal handling of mmap syscall? I suspect the EINVAL is possibly because kernel may have linked the sysfs entry to a char device that does not implement mmap? (but I do not know for sure). 

Any comments or ideas? Did anything change?

Regards,
Michal

michkrom, 2025-04-01:
Further information:
- repeatable on Arch/Manjaron (tried 6.6 and 6.13 kernels)
- not repeatable on Ubuntu (tried 6.8 and 6.11 kernels)

So likely kernel config options?

michkrom, 2025-04-04:
Indeed, it was CONFIG_STRICT_DEVMEM. Apparently, Arch kernel is compiled with it but Ubuntu is not.

hmaarrfk, 2025-04-04:
did you find a way to get it to be compatible with `CONFIG_STRICT_DEVMEM`

michkrom, 2025-04-04:
Well, yes. In my yokto I had to disable CONFIG_STRICT_DEVMEM as it was on by default. For Arch, you'd have rebuild the kernel yourself. Alternatively, instead of mmaping via sysfs, you can map via /dev/h2c etc created by the xdma driver.

hmaarrfk, 2025-04-04:
>  Alternatively, instead of mmaping via sysfs, you can map via /dev/h2c etc created by the xdma driver.

I see this was the kind of "pointer" i was looking for.

Thanks for sharing!

- Your friend in this lack of support from Xilinx!